nrwiersma/ConfigManager #1: Librery

shaddow501, 2016-08-18:
Dear Sir

Have you tried to install this example on a clean nodemcu?
I have spent hours trying to understand the problem but you see I am lost... So I write to you.

I have uploaded the code and try to connect to the "demo" AP trough my mobile, it opens the connection manager address http://192........
and I get file not found...

I have tried to understand as much as I can but I am lost, it seems that it doesnt upload the index.html...

Can you check the code again please?

I am also curious, you say that the SSID and password is saved in the eeprom, what if after that I connect to one access point I would like to connect to another? should I reprogram again the code or is there other way to reset settings?

nrwiersma, 2016-08-19:
Hi,

It seems I have a piece missing from the documentation that I will go and fix after this. The index.html needs to be uploaded to the SPIFFS. Your IDE and setup will determine how this is done. The instructions for Arduino can be found here: [Uploading files to file system](https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system). I have provided an index.html for you as an example in the ```data``` directory, but you are welcome to change it if you like, as long as the data gets posted in the same format.

With regards to the saved SSID, the configuration will try to connect to its stored SSID, if it fails to do so, it will then start the captive portal so you can set a new SSID and password. If the access points are not in range of each other, then you should not need to reflash. If your use case is that they are next to each other, I could possibly expose a REST endpoint to set the new SSID and password. Please let me know if you need this?

Thanks,
Nicholas Wiersma

nrwiersma, 2016-08-20:
Hi,

Did this help? Does it resolve your issue?

shaddow501, 2016-08-20:
Dear Sir
Thanks for your fast responce, Yes it did help but I moved on to other library that also show me the access points I can chose from and reset configuration if I would like to change the access point (that is without re-flashing the code, the library name is Wifimnager)
Anyway in your Simple.ino example this argument is missing:  ConfigManager configManager;
Kind regards,
Ilan.

nrwiersma, 2016-08-20:
Thanks, will check up on that and update the library when I get a chance.

closed 2016-08-20 by nrwiersma