nrwiersma/ConfigManager #4: no matching function for call to 'ConfigManager::addParameter(const char [5], char (*)[20], int)

ovi1337, 2016-09-29:
Hi i'm using Platformio and would like to use your library for an ESP8266, but i can't use addParameter for strings, because i will get this error:
```
src/main.cpp:15:56: error: no matching function for call to 'ConfigManager::addParameter(const char [5], char (*)[20], int)'
```

Here is the whole error log of the compiling:
```
src/main.cpp:15:56: note: candidates are:
In file included from src/main.cpp:1:0:
lib/ConfigManager/ConfigManager.h:107:10: note: template<class T> void ConfigManager::addParameter(const char*, T*)
void addParameter(const char *name, T *variable) {
^
lib/ConfigManager/ConfigManager.h:107:10: note:   template argument deduction/substitution failed:
src/main.cpp:15:56: note:   candidate expects 2 arguments, 3 provided
configManager.addParameter("name", &config.name, 20);
^
In file included from src/main.cpp:1:0:

lib/ConfigManager/ConfigManager.h:110:10: note: void ConfigManager::addParameter(const char*, char*, size_t)
void addParameter(const char *name, char *variable, size_t size) {
^
lib/ConfigManager/ConfigManager.h:110:10: note:   no known conversion for argument 2 from 'char (*)[20]' to 'char*'
*** [.pioenvs/nodemcuv2/src/main.o] Error 1
```

If i'm commenting out the line `configManager.addParameter("name", &config.name, 20);` can i compile and use your library without any problems. But then i can't use strings/chars :(

Here is my __main.cpp__
```
#include "ConfigManager.h"

struct Config {
    char name[20];
    bool enabled;
    int8 hour;
} config;

ConfigManager configManager;

void setup() {
    // Setup config manager
    configManager.setAPName("Demo");
    configManager.setAPFilename("/index.html");
    configManager.addParameter("name", &config.name, 20);
    configManager.addParameter("enabled", &config.enabled);
    configManager.addParameter("hour", &config.hour);
    configManager.begin(config);

    //
}

void loop() {
    configManager.loop();
}
```

Can you help find out the problem and/or are you be able to help me for this problem?

nrwiersma, 2016-09-29:
Hi,

What version of the library do you have installed? To use strings you need to be using 0.5.

Let me know if this isn't the issue.

ovi1337, 2016-09-29:
Hi,
i'm using the current master from github

nrwiersma, 2016-09-29:
Ah,

I worked it out. Change this line

```
configManager.addParameter("name", &config.name, 20);
```

to this

```
configManager.addParameter("name", config.name, 20);
```

I will update the example and documentation. Sorry about that.

ovi1337, 2016-09-29:
Awesome, I will test it in some minutes, thank you very much :)

nrwiersma, 2016-09-30:
Hi,

Was wondering if this solved your issue?

ovi1337, 2016-09-30:
Hi, 
yes, all is fine now - thank you :) 
I will close the issue now.

closed 2016-09-30 by ovi1337