nrwiersma/ConfigManager #6: AP remains available after connecting node to wifi

robertoostenveld, 2016-11-06:
I noticed this with an NodeMCU ESP8266 development board. I consider it undesired behaviour that the AP remains available, as it allows anyone to reconfigure the Thing. Let me outline the steps that allow me to reproduce.

When the Thing is not connected to my local wifi network, I see “Starting Access Point” in the serial console and can connect my laptop as follows:

Laptop Wi-Fi is connected to ESP-THING and has the IP address 192.168.1.100.

All is fine and I configure the SSID and Password of my wifi router. In the serial console I can see it connecting. The ESP-THING wifi network shortly disappears, and then reappears. With the Thing  connected to my local wifi network, I can again connect to its AP as follows:

Laptop Wi-Fi is connected to ESP-THING and has the IP address 192.168.4.2.

In the latter case (i.e. with the thing connected), I can access the /index.html endpoint at 192.168.1.13 (when I connect my laptop to my router network) and at 192.168.4.1 (when I connect my laptop to the ESP-THING network).

What I do not understand is where the 192.168.4.x network comes from, as there is nothing in the code with this IP address. There is this

````
    IPAddress ip(192, 168, 1, 1);
    WiFi.mode(WIFI_AP);
    WiFi.softAPConfig(ip, ip, IPAddress(255, 255, 255, 0));
    WiFi.softAP(apName);
````

I do not find in the code where the mode is being switched, but I get the feeling that - once connected - it is in WIFI_AP_STA and not in WIFI_STA mode.

nrwiersma, 2016-11-06:
Hi,

Thanks for looking into this. It is something I had at some point noticed, but thought it an artifact.
I am working on another issue today, so will include your change in my next update version.

Thanks.

closed 2016-11-06 by nrwiersma

nrwiersma, 2016-11-06:
This fix has been included in the 0.7 release. Thanks for contributing.

robertoostenveld, 2016-11-06:
thanks, I'll update to your latest version.