nrwiersma/ConfigManager #9: configuration parameters other than int don't work

robertoostenveld, 2017-01-07:
I have noticed in my own sketch but also in the example/simple.ino that the "name" (string) and "enable" (boolean) cannot be updated through a PUT call

This is immediately after uploading the sketch

```
mbp> curl -X GET http://192.168.1.16/settings
{"name":"","enabled":true,"hour":0}
```

Here I am trying to update the three config options

```
mbp> curl -X PUT -d '{"name":"something","enabled":false,"hour":2}' http://192.168.1.16/settings
```

But only the hour (int) changes.

```
mbp> curl -X GET http://192.168.1.16/settings
{"name":"","enabled":true,"hour":2}
```

The work-around that I am now using is to only use integers in the configuration.

nrwiersma, 2017-01-07:
Hi,

I am having trouble duplicating your results, although I think it might have something to do with having an empty EEPROM. In mine I can flash both name and enabled.

In one of my sketches, i run ```configManager.save()``` in setup, perhaps being why I have never seen this. Could you give this a try? perhaps it will work for you?

Let me know and thanks for letting me know.

robertoostenveld, 2017-01-10:
I'll try it at the next occasion. 
I realize that it might also be because of having another version of ArduinoJson.

nrwiersma, 2017-01-10:
True, that could also be the case.

nrwiersma, 2017-02-01:
I assume you got this sorted. I will close this issue. Please reopen it if you see the issue again.

Thanks.

closed 2017-02-01 by nrwiersma