nrwiersma/ConfigManager #12: Nice but the example doesn't work.

nardev, 2017-08-06:
I just compiled/uploaded this to a ESP8266, it boots, makest AP, i connected but no way to get in.
Expected urls don't work..

nrwiersma, 2017-08-06:
Hi,

Can you give more information please. It is difficult to know what is going on from your description.

Thanks

nrwiersma, 2017-08-23:
Going to close this due to lack of activity, if you still having an issue, please re-open and I will gladly help where I can.

closed 2017-08-23 by nrwiersma

menan, 2018-03-04:
I faced this issue. 

when I call `configManager.setAPICallback(createCustomRoute);` before `configManager.begin(config);` instead of after, it worked. Thanks the library @nardev.

nrwiersma, 2018-03-04:
Thanks, you are correct, if it starts the API really fast, this might be the case. I will move it before in the example. Thanks for reporting.