nrwiersma/ConfigManager #14: Clear-text passwords exposed in GET api

JonSilver, 2017-08-17:
The GET api passes the contents of all custom field to the client in clear text. Any custom fields which carry sensitive information, e.g. MQTT broker passwords, are therefore easy security breach targets.

I previously suggested:
1. implement an optional flag on custom fields which requests that some sort of two-way encryption/hashing be used to store and retrieve the custom field
2. the GET api doesn't bother decrypting these fields before sending to the client

However, I think it might be sufficient to still store password fields in clear text but send them obscured by filling the field's buffer with asterisks *.

How do you feel about that Nick?

Continued from previous discussion in issue #11

nrwiersma, 2017-08-17:
Hi,

I like the suggestion of the asterisks. Another idea is to add a new function ```addSetParameter``` that does not return the parameter in the GET at all. How does that sound?

Thanks,
Nick

JonSilver, 2017-08-18:
That sounds like a very good idea, Nick.

nrwiersma, 2017-08-22:
Got this done, let me know what you think.

PR #17 (2017-08-22): Parameter modes
Fixes #14
closed 2017-08-22 by nrwiersma