nrwiersma/ConfigManager #21: ESP32 Support

JonSilver, 2018-06-22:
Hi Nick, have you considered adding ESP32 support for ConfigManager? Happy to help out once again if you have the time.

nrwiersma, 2018-06-25:
So after looking into this, it seems that a compatible WebServer is missing from the ESP32 library, although there does seem to be a PR [#430](https://github.com/espressif/arduino-esp32/pull/430) to add it in. Once this has been added, it should not be too difficult to adding support.

naveenneog, 2018-07-02:
They have closed PR#430 and It is available in the build. I tried to compile the ConfigManager for ESP32 using WebServer and Wifi libraries, I hit a wall with ArduinoJson. @nrwiersma you would be the best person to port it. It would be great If you could compile it for ESP32.

nrwiersma, 2018-07-02:
@naveenneog according to the docs, it should be [compatible](https://github.com/bblanchon/ArduinoJson).

nrwiersma, 2018-07-08:
@naveenneog So it seems you have 2 issues there: ArduinoJson and SPIFFS. Not sure why it cannot find the libraries though, it depends on your setup as well. I will give compiling this a shot soon.

nrwiersma, 2018-07-08:
I have started a branch [esp32](https://github.com/nrwiersma/ConfigManager/tree/esp32) that has the correct bindings. Still having a couple build issues. I will work on it.

JonSilver, 2018-07-08:
I'm sorry I've not been available to get involved... in the midst of looming project deadlines. Looks like great progress is being made anyway. Thank you @nrwiersma and @naveenneog.

nrwiersma, 2018-07-08:
@JonSilver @naveenneog So [the branch](https://github.com/nrwiersma/ConfigManager/tree/esp32) is now compiling for me, but I still have no device to test this on. Can either of you confirm this is working?

nrwiersma, 2018-07-08:
@naveenneog I think your compiler cannot find ArduinoJson library. I fixed the SPIFFS not being included issue in my branch.

naveenneog, 2018-07-09:
@nrwiersma  I tried to build it for ESP32 , It builds fine now. I had to roll back the ArduinoJson library to older version(https://github.com/bblanchon/ArduinoJson/issues/756) .
However ConfigManager.reset() is been removed it seems in your new update, my old code is giving error for that.

nrwiersma, 2018-07-09:
@naveenneog Sorry, I am confused? There is no reset function, nor has there ever been a reset function. I had to check, but i have never written one. What did it do?

naveenneog, 2018-07-10:
My mistake , I was referring to the forked library. I successfully compiled and burnt it to my ESP32.

nrwiersma, 2018-07-10:
@naveenneog Awesome, did it work as expected? @JonSilver Could you perhaps confirm this success? Would love to get this merged and tagged.

bblanchon, 2018-07-11:
You can verify that the right version of ArduinoJson is installed by checking the [ARDUINOJSON_VERSION](https://arduinojson.org/v5/api/utils/version/?utm_source=github&utm_medium=issues) macros.

nrwiersma, 2018-07-18:
So if there are no objections, I will merge this change today? @bblanchon this is a great idea, don't know what version we need. Perhaps we can make this a separate pull request?