nrwiersma/ConfigManager #25: 'class ConfigManager' has no member named 'reset'

dsokac, 2018-08-18:
`configManager.reset();`
The version I am using is: 1.2.0

nrwiersma, 2018-08-18:
I am confused by the question. ConfigManager does not has a reset function. Where did you find the example code that includes it?

dsokac, 2018-08-18:
https://ubidots.com/blog/connect-your-esp8266-to-any-available-wi-fi-network/

I used their code... is this anyway needed?

dsokac, 2018-08-18:
Thank you for quick responses :) wish you all the best, and let me know if you find out something I may want to know.

closed 2018-08-18 by dsokac

nrwiersma, 2018-08-20:
Hi,

Just to let you know, it seems the article points to a fork of my code (a friend pointed this out, I completely missed it). It can be found here:
https://github.com/mariacarlinahernandez/ConfigManager/tree/maka/ConfigManagerForked

Thanks,
Nick