nrwiersma/ConfigManager #31: Secure the AP with PASSWORD

bigoper, 2018-09-23:
Thank you for an amazing Library!

Scenario:
- The device is now in AP mode.

Issue:
- The AP is not secured

Request:
- Is it possible to secure the AP with a PASSWORD?

Thanks!

nrwiersma, 2018-09-23:
No worries. This goes against most use cases, but it should be fairly straight forward to support. If I make a branch for this, would you be able to test it for me?

bigoper, 2018-09-24:
Sure, with pleasure.
Just let me know when.

Many thanks!

bigoper, 2018-09-25:
> Hi,
>
> Awesome, thanks. Can you give this branch a try?
> https://github.com/nrwiersma/ConfigManager/tree/ap-password
>
> Thanks.

Sure, I'll do it today.
I'll post the results ASAP.

Thanks!

aviivgi, 2018-09-25:
I've tested the new branch -> works like a charm!
Thank you so much for the quick response/update.

Keep doing what you're doing :)

bigoper, 2018-09-25:
I've tested the new branch -> works like a charm!
Thank you so much for the quick response/update.

Keep doing what you're doing :)

closed 2018-09-25 by bigoper

nrwiersma, 2018-09-26:
Awesome, glad it works. I will then merge this into master and tag a new version. After that the branch will be deleted.

commit 5c6aedea (2018-09-26): Add AP password (#32)

Fixes #31