nrwiersma/ConfigManager #38: example REST to save and retrieve JSON

emitterpark, 2019-01-01:
Hi, 
I am new to programming. Thank you for the configmanager.

I want to save/retrieve ESP wifi config and additional parameters from my JS mobile app fetch api REST. 

Uploaded simple.ino example sketch.
Tested in my windows machine, curl, this works for config wifi credentials; ESP can connect to my home AP.
Curl -X POST http://192.168.1.1 -d “ssid=my_ssid&password=my_password”

While reconnected to “Demo” AP, This way to save parameters;
curl -H "Content-Type: application/json" -X PUT http://192.168.1.1/settings -d “{here JSON for various types}" 

Then this way to see response in console; (always blank)
Curl -X GET http://192.168.1.1/settings 

I didn't understand well the structure for various parameter types in sketch OR a wrong syntax in curl lines.
Could you please give an example to save and retrieve a string parameter, “key”:”value”.
Both in sketch and curl statements.

Thanks

nrwiersma, 2019-01-05:
Hi,

Been really busy over the last period. What about the parameters do you not understand?

If you are using the `simple.ino` sketch, then the example in the readme for PUT should work. This is how I test it. The "key" should be the name (first parameter) given to the parameter. The second is the pointer to the variable that will be set. In the case of a string parameter there is a 3rd parameter which is the length of the `[]char`. There are other options, but you dont need to worry about them for now.

I stuff is packed away at them moment, and wont be able to get it out for a couple days. If you would like to try and post your code here, I can have a look at it, otherwise i will try make something up for you in a bit.

mitchsf, 2019-01-08:
I too would like to see an example to GET a value. I've successfully run the sample and added a few fields to index.html and it works, but I'm not sure how to use endpoints. Just a few lines of code to demonstrate is all I need.

Great library by the way, just what I need. Thanks.

nrwiersma, 2019-01-08:
As soon as I get a chance, I will post some code. Can you tell me what would help with regards to the README to clear this up?

mitchsf, 2019-01-08:
Thanks, no rush. If you would add a few more fields to index.html, then retrieve them to variables after "save" is clicked, that's all I need. I will ultimately add around 30 fields to input data, all either text or dropdowns.

nrwiersma, 2019-01-08:
That isnt the actual purpose of the `index.html`. Its purpose is mainly to connect to the Wifi network. It was always the goal that the settings should be managed from an external page, hence the REST style settings get/put. The way I have used it is to do a GET first to "discover" the parameters and values, and build the settings form on the fly. Then updates can be PUT back in.

mitchsf, 2019-01-08:
Thanks for the explanation. If you have time, an example would really help.

emitterpark, 2019-01-09:
Without uploaded index.htmI to SPIFSS,
tried discovering paths below and have HTTP/1.1 404 Not Found.

`curl -i -H "Content-Type:application/json" -X GET http://192.168.1.1/settings
 curl -i -X GET http://192.168.1.1/settings`

Or

`curl -i -H "Content-Type:application/json" -X GET http://192.168.1.1/custom
 curl -i -X GET http://192.168.1.1/custom`

But the root works for wifi discover or save.
`Curl -X POST http://192.168.1.1 -d “ssid=my_ssid&password=my_password”`

baphled, 2019-02-22:
I've been stumped by this too, mainly as the callback wasn't working with the sample code. I've finally got something working so I'm more than happy to add it to the example, if it's of use to others.

nrwiersma, 2019-02-22:
Sorry, i have not had time to get to this. Could you perhaps include a second example with your code? Under a new folder under `examples`. That would be greatly appreciated.

baphled, 2019-02-22:
@nrwiersma, no worries, I know how that goes.

Sweet, that works for me, I'm just doing some clean up but I'll push something today or other the weekend 👌

nrwiersma, 2019-02-23:
Fixed by #41.

closed 2019-02-23 by nrwiersma