nrwiersma/ConfigManager #42: ArduinoJson 6 incompatibility

JonSilver, 2019-03-30:
The library should either be updated to be compatible with breaking changes in ArduinoJson 6, or the documentation augmented to advise that ArduinoJson 5 be used.
In the meantime, for anyone reading this who uses PlatformIO, this can be done by adding the following to `platformio.ini`:

    lib_deps =
        ArduinoJson@5.13.4

nrwiersma, 2019-03-30:
Thanks. I have done as you suggested and stated in the library.json and the readme that version 5 is used. Version 6 has a fair number of changes.

closed 2019-03-30 by JonSilver

rishabhrkaushik, 2019-04-07:
Any plans for migrating to ArduinoJSON 6 in near future?

nrwiersma, 2019-04-08:
Not right now. Did look at it briefly, but it is a large break between the versions, so will have to figure out how to use it again.

JonSilver, 2019-04-08:
In the meantime @nrwiersma if I happen to get there first then I'll let you know and submit a PR.

nrwiersma, 2019-04-08:
Thanks, always appreciate the help.

ReneHezser, 2020-04-01:
Just wanted to check if there is any progress in the port to V6.

nrwiersma, 2020-04-02:
@ReneHezser It is not a focus of mine at the moment. If you would like to contribute the conversion that would be great.

ReneHezser, 2020-04-02:
I did :-)
My tests with ESP8266 and ESP32 with both examples succeeded.