nrwiersma/ConfigManager #46: save-config-demo

RudyFiero, 2019-11-11:
First I want to say that I like your library. It does what it does without a lot of junk in the way. What I want is a clean way to set the WiFi connection, and to set some other parameters to eeprom. I tried another library but it was doing too much. Has it's own way to make web pages that was "simpler". I thought it only made it harder as I needed to figure out how to work with it. 

The minor issue I have with ConfigManager is on a new setup the contents of the eeprom is filled into the the fields in the setup screen. I haven't looked at the code yet, but it would be better to have empty text boxes on the first load. Typing new values into the line does not get rid of the remainder of the line.

I just tried to save. I think it might have saved. But I don't know if it did. Confirmation would be nice. I guess I need to handle it myself. Hey, at least I can do it my way.

markvriens, 2019-11-11:
I had the same problem.

nrwiersma, 2019-11-12:
Hi,

> The minor issue I have with ConfigManager is on a new setup the contents of the eeprom is filled into the the fields in the setup screen. I haven't looked at the code yet, but it would be better to have empty text boxes on the first load.

The system has no concept of "first load". If your EEPROM is zeroed when flashing, then the values will be zeroed out. To try and change behaviour for first load would just make things more complex, as we would need to keep track of the state of the device. If it was desired not to read config that had not yet been written, it would be possible to add config magic bytes. When reading, if the bytes are not seen, it would not bother reading the config. This would allow for "default" config to be set on startup.

> Typing new values into the line does not get rid of the remainder of the line.

Could you explain this more, I dont understand the issue here. Are you talking in the browser?

Thanks,
Nick

RudyFiero, 2019-11-15:
[image]

I did an erase of all flash and I then got what is shown in the image above.  If the contents of the EEPROM area is filled with FF (erased) would that not give you a concept of "first load".

When I entered the new values into the form, the entered data wrote into those character positions but the remainder of the line was still filled with the ? image. 

If this was meant to be just for myself then I wouldn't have a problem with the quirks. But whatever I develop, I want to be able to give this to 'normal' people.

nrwiersma, 2019-11-15:
The issue here is that you are erasing with `FF` and not `00`. With strings in C, they are `00` terminated, so if you erase with `00` it will auto terminate the string. I kinda assumed erasing with `00` was standard.

RudyFiero, 2019-11-15:
When I start with a clean slate, select erase "all flash contents" in the menu, any not initialized location will have FF in it. When you perform an erase on a memory block the erased state is all ones, FF. When you write some data what is happening is the appropriate bits are made to be zeros. 

> With strings in C, they are 00 terminated,

Then should the structure be initialized with 00 ? 

Anyway, while there can be improvements, this is a good starting point for me. Thanks

closed 2019-11-15 by RudyFiero

cgmckeever, 2019-11-25:
@RudyFiero FWIW in the latest release there is a `clearSettings` call which should (maybe) initialize the memory that is handled more consistently per this discussion

RudyFiero, 2019-11-25:
Great! I will give it a try as soon as I can.