nrwiersma/ConfigManager #61: What is the point of "enabled" in config examples?

canadaduane, 2019-11-24:
I don't see this parameter being used anywhere, but it is offered in both the `simple.ino` and `save_config_demo.ino` examples:

```
struct Config {
  char name[255];
  bool enabled;
...
```

What is "enabled" intended for?

nrwiersma, 2019-11-24:
It is just example configuration. Your configuration can be what ever you need it to be.

cgmckeever, 2019-11-24:
so .. I had this issue onboarding as well ... as @nrwiersma said, they are for your configuration that you are managing, not for the utility itself ... There is a note in the README to clear this up, but maybe it needs to be teased out some more?

```
Note: These are examples of possible settings. The config is required, these values are arbitrary.
```

canadaduane, 2019-11-25:
Thanks! One thing that would have helped me understand this earlier is to
see HOW to use these config variables, i.e. in the loop() function
somewhere. Then it would be clear that these are "things I can do stuff
with" (and not "things ConfigManager needs").

closed 2019-12-28 by nrwiersma