nrwiersma/ConfigManager #64: Remove jQuery from Settings Example

canadaduane, 2019-11-28:
The `save_config_demo` example has a [settings.html](https://github.com/nrwiersma/ConfigManager/blob/master/examples/save_config_demo/data/settings.html) file that uses a CDN-hosted jQuery to update the form and post form results back to the ESP/ConfigManager API.

Currently, requiring CDN-hosted jQuery makes for a less-than-ideal experience. In some cases, internet is not available, and it would be unfortunate if that meant additional settings could not be set between a user and their device.

nrwiersma, 2019-11-28:
Hi,

To be clear, this is a demo. What you put on your actual device is entirely up to you. If you are working in a disconnected situation, this is not ideal, but as an example, it is clear and concise.

canadaduane, 2019-11-29:
I think I just realized there's an assumption built in to settings that I
didn't consider. Is it necessary to connect to WiFi before modifying the
custom settings?

cgmckeever, 2019-11-29:
As the examples are set up YES

 It you can add the custom settings route to the APCallback hook to allow
that route to be available in AP or API MODE

cgmckeever, 2019-11-30:
@canadaduane example:

```
void APCallback(WebServer *server) {
    server->on("/styles.css", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(stylesCSS, mimeCSS);
    });

    // configure settings route (note: example requires js/css includes)
    server->on("/settings.html", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(settingsHTML, mimeHTML);
    });

    // NOTE: css/js can be embedded in a single page HTML 
    server->on("/styles.css", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(stylesCSS, mimeCSS);
    });

    server->on("/main.js", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(mainJS, mimeJS);
    });
}
```

canadaduane, 2019-11-30:
I think what I most want to be able to do is create an `index.html` file that has both SSID/password AND custom settings. If I'm not mistaken, with the current architecture, I MUST have javascript enabled to do this, since I have to hit two endpoints (one for wifi SSID/pass to go to AP and one for custom settings to go to API), correct?

This seems like more complication than I'd like, but maybe I'm missing something. Why not allow custom settings to be POSTed back to the same form that accepts SSID/password?

cgmckeever, 2019-12-01:
You can def add a POST route for settings and not use JS

You can also merge both forms for config and WIFI into one. Where it gets
tricky is bubbling errors back up to the user — where the JS allows the
client side validation and you can handle error replies via Json

But everything you asked of def provided in the core framework — you’ll
just need to write some custom POST handlers

canadaduane, 2019-12-01:
So, something like this? What would I do at `do something with playerName?`?

```
void APCallback(WebServer *server) {
    server->on("/update", HTTPMethod::HTTP_POST, [server](){
        String ssid, password, playerName;
        ssid = server->arg("ssid");
        password = server->arg("password");
        playerName = server->arg("player_name");

        if (ssid.length() == 0) {
            server->send(400, FPSTR(mimePlain), F("Invalid ssid."));
            return;
        }

        storeWifiSettings(ssid, password, false);
        
        // do something with playerName?

        server->send(204, FPSTR(mimePlain), F("Saved. Will attempt to reboot."));
        ESP.restart();
    });
}
```

IceBearZero, 2019-12-01:
Yes that pattern would work. IMHO - I would at least make a separate settings page, as I find it odd that the WIFI/etc is all wrapped next to the playersName, but then again its only 1 additional field .. In addition, the last line for that call does restart the system, which may not be ideal for your case

closed 2019-12-28 by nrwiersma