nrwiersma/ConfigManager #68: Request! Single Page web interface

te260701, 2019-12-25:
Hi,

I think single page would be better for user experience since no need to remember to type .../settings.html to get to config page.

Unless, I missing something. After connect to wifi network, I type local IP it still show Wifi Details page (index.html). I have to type settings.html to get to config.

I see you have AP and API mode. Why dont we just show settings page by default when we are in API mode.

I think passing mode: ap or api to json data and using this mode to hide/show wifi/settings div on single page. I have no idea how to edit the code to pass this mode to browser.

I also try to change configHTMLFile = "/index.html" and rename settings.html to index.html hoping it will show settings page be default but it still show wifi details page which I rename it to wifi.html which is weird. It seems it doesn't matter which mode web server will start in ap mode (I set APFilename=wifi.html)

Thank you in advance,
Sam

```
#include "ConfigManager.h"

const char mimeHTML[] PROGMEM = "text/html";
**const char *configHTMLFile = "/index.html";**

struct Config {
  char name[255];
  bool enabled;

  char mqtt_host[255];
  char mqtt_port[10];
  char mqtt_username[255];
  char mqtt_password[255];
  char mqtt_topic[255];
} config;

struct Metadata {
  int8_t version;
} meta;

ConfigManager configManager;

void createCustomRoute(WebServer *server) {
  **server->on("/", HTTPMethod::HTTP_GET, [server](){**
    SPIFFS.begin();

    File f = SPIFFS.open(configHTMLFile, "r");
    if (!f) {
      Serial.println(F("file open failed"));
      server->send(404, FPSTR(mimeHTML), F("File not found"));
      return;
    }

    server->streamFile(f, FPSTR(mimeHTML));

    f.close();
  });
}

void setup() {
  Serial.begin(115200);
  Serial.println("");

  meta.version = 3;

  // Setup config manager
  configManager.setAPName("Demo");
  **configManager.setAPFilename("/wifi.html");**

  configManager.addParameter("name", config.name, 255);

  configManager.addParameter("mqtt_host", config.mqtt_host, 255);
  configManager.addParameter("mqtt_port", config.mqtt_port, 10);
  configManager.addParameter("mqtt_username", config.mqtt_username, 255);
  configManager.addParameter("mqtt_password", config.mqtt_password, 255);
  configManager.addParameter("mqtt_topic", config.mqtt_topic, 255);

  configManager.addParameter("enabled", &config.enabled);
  configManager.addParameter("version", &meta.version, get);

  configManager.setAPICallback(createCustomRoute);

  configManager.begin(config);
  Serial.print("ESP MODE:");
}

void loop() {
  configManager.loop();
}
```

IceBearZero, 2019-12-25:
```
Unless, I missing something. After connect to wifi network, I type local IP (http://192.168.1.137) it still show Wifi Details page. I have to type settings.html to get to config.
```
I think you can change this to default to whichever page you want as root, as it currently is .. 

There was an earlier issue/discussion about some thoughts on why its two separate pages ... There was some reason, not set in stone, but made sense

te260701, 2019-12-25:
Hi IceBearZero,
Thank you for quick response,
I did change API callback to root. 
```
void createCustomRoute(WebServer *server) {
  server->on("/", HTTPMethod::HTTP_GET, [server](){
```
But it doesn't work. 

It must be somewhere else. Could you show me where?

cgmckeever, 2019-12-25:
https://github.com/nrwiersma/ConfigManager/blob/master/src/ConfigManager.cpp#L344
This does the initial set of the `/` route.

I'd be surprised if it doesnt let you overwrite it.
Have you 

```
const char *settingsHTML = (char *)"/settings.html";

void createCustomRoute(WebServer *server) {
  server->on("/", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(settingsHTML, mimeHTML);
    });

```

closed 2019-12-26 by te260701

IceBearZero, 2019-12-26:
Did this work out?

te260701, 2019-12-26:
Hi @IceBearZero 

I finally get it to work by modify configmanager cpp and h file. Now when API mode, it show settings page by default. I add /wifi to get to wifi setting page or still can access by apFilename (../index,html). 

add handleAPIGet()
```
void ConfigManager::handleAPIGet() {
    streamFile((mode == api ? apiFilename : "/settings.html"), mimeHTML);
}
```

add setAPIFilename() and no need to have settingHTML variable in sketch
```
void ConfigManager::setAPIFilename(const char *filename) {
    this->apiFilename = (char *)filename;
}
```

edit createBaseWebserver()
```
void ConfigManager::createBaseWebServer() {
    const char* headerKeys[] = {"Content-Type"};
    size_t headerKeysSize = sizeof(headerKeys)/sizeof(char*);

    server.reset(new WebServer(this->webPort));
    DebugPrint(F("Webserver enabled on port: "));
    DebugPrintln(webPort);

    server->collectHeaders(headerKeys, headerKeysSize);

	/* te260701 change this line  
    server->on("/", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleAPGet, this));
	*/
	if (mode==api) {
		server->on("/", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleAPIGet, this));
	} else { 
		server->on("/", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleAPGet, this));
	}
	/* end of te260701 change */
    server->on("/", HTTPMethod::HTTP_POST, std::bind(&ConfigManager::handleAPPost, this));
    server->on("/scan", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleScanGet, this));
    server->onNotFound(std::bind(&ConfigManager::handleNotFound, this));	
}
```

Edit startapi()
```
void ConfigManager::startApi() {
    mode = api;

    createBaseWebServer();
    server->on("/settings", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleRESTGet, this));
    server->on("/settings", HTTPMethod::HTTP_PUT, std::bind(&ConfigManager::handleRESTPut, this));
	/* add by te260701 */
	server->on(apiFilename, HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleAPIGet, this));
	server->on("/wifi", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleAPGet, this));
	server->on(apFilename, HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleAPGet, this));
	/* end add by te260701 */

    if (apiCallback) {
        apiCallback(server.get());
    }

    server->begin();
}
```

te260701, 2019-12-27:
> https://github.com/nrwiersma/ConfigManager/blob/master/src/ConfigManager.cpp#L344
> This does the initial set of the `/` route.
>
> I'd be surprised if it doesnt let you overwrite it.
> Have you
>
> ```
> const [...]

Hi @cgmckeever 
Sorry, I missed your answer. It seems it take whatever first matched route found, in this case handleAPGet instead of customRoute one.

IceBearZero, 2019-12-27:
If you move this line (and/or its correspnding in AP setup) above any of the other route configs, then the custom will be created first .. 

https://github.com/nrwiersma/ConfigManager/blob/master/src/ConfigManager.cpp#L339

te260701, 2019-12-27:
@IceBearZero 
Do you mean move it before Line 333: createBaseWebServer() ?

IceBearZero, 2019-12-27:
yes, in either call (AP or API) if you move the custom routes before any other routing call (ie `createBaseWebServer` then if the routes do not get reroute, the custom routes should take the precedence

te260701, 2020-01-02:
> yes, in either call (AP or API) if you move the custom routes before any other routing call (ie `createBaseWebServer` then if the routes do not get reroute, the custom routes should take the precedence

I got boot loop when I move `createBaseWebServer` to after `apiCallback`. It works fine in AP mode but got boot loop after enter SSID and Password.
 
```
void ConfigManager::startApi() {
    mode = api;

    server->on("/settings", HTTPMethod::HTTP_GET, std::bind(&ConfigManager::handleRESTGet, this));
    server->on("/settings", HTTPMethod::HTTP_PUT, std::bind(&ConfigManager::handleRESTPut, this));

    if (apiCallback) {
        apiCallback(server.get());
    }

    createBaseWebServer();
    server->begin();
}
```