nrwiersma/ConfigManager #69: Cannot save config bool on v1.5.1

te260701, 2019-12-26:
Hi there,

Merry Christmas :)
I just upgrade to v1.5.1 and try to add bool parameter and int parameter to config but it bool param won't save, int param OK.

EDIT: I found issue with main.js which is not create json data properly.
Here the code I edit. (I'm not expert in anyway so, wait for developer to come up with better solution lol)

```
  $.fn.serializeObject = function() {
    var o = {};
    var a = this.serializeArray();
    $.each(a, function() {
      var input = document.getElementsByName(this.name);
      var value = this.value;
      var dataType = input[0].getAttribute("data-type");
      if (dataType == "number") value = parseFloat(value);
      if (dataType == "bool") {
		  value = (value=="true" || value==1)? true:false;
      }  
      if (o[this.name]) {
        if (!o[this.name].push) {
          o[this.name] = [o[this.name]];
        }
        o[this.name].push(value || '');
      } else {
        o[this.name] = (value==null) ? '' : value;
      }
    });
    return o;
  };

```

IceBearZero, 2019-12-26:
Yeah, this is one of the reasons that `data-type` attribute was in there. There is probably a more elegant way to do this in JS, but this works ;)

closed 2020-01-02 by nrwiersma

te260701, 2020-01-02:
@nrwiersma 
It does correctly save data but it does not have JS to set value (checked) for boolean checkbox.

nrwiersma, 2020-01-03:
@te260701  This is now fixed.

PR #70 (2020-01-02): Add boolean config example
This PR adds a boolean example for saving config.

Fixes #69