nrwiersma/ConfigManager #75: Saving credentials does not work

ReneHezser, 2020-03-14:
Thanks for this great library. I am trying to get it running on an ESP32 via PlatformIO.
In the log I can see this output during boot and when I try to save my Wifi credentials.

> Starting Access Point
AP Name: SolarPowerReader
AP IP address: 192.168.1.1
Webserver enabled on port: 80
dhcps: send_offer>>udp_sendto result 0
[E][WebServer.cpp:617] _handleRequest(): request handler not found
Unknown URL: www.msftconnecttest.com
_--> at this point I hit save in the browser to store the credentials <--_
WiFi Settings cannot be stored before ConfigManager::begin()
ets Jun  8 2016 00:22:57

The code for testing:

```cpp
#include <Arduino.h>
#include "ConfigManager.h"

struct Config
{
  char name[20];
  bool enabled;
  int8_t hour;
  char password[20];
} config;
struct Metadata
{
  int8_t version;
} meta;

ConfigManager configManager;
const long interval = 30000;
// the logic should be triggered on the first run and not after the interval
unsigned long previousMillis = millis() - interval;

void setup()
{
  DEBUG_MODE = true; // will enable debugging and log to serial monitor

  Serial.begin(115200);
  Serial.println("\n\nbooting...\n\n");

  meta.version = 3;

  // Setup config manager
  configManager.setAPName("SolarPowerReader");
  configManager.setAPFilename("/index.html");
  configManager.addParameter("name", config.name, 20);
  configManager.addParameter("password", config.password, 20, set);
  configManager.addParameter("version", &meta.version, get);

  configManager.begin(config);
  // configManager.clearWifiSettings(false);
  //configManager.save();
}

void loop()
{
  configManager.loop();

  unsigned long currentMillis = millis();
  // query Modbus if interval is reached and Wifi is connected
  if (currentMillis - previousMillis >= interval && configManager.getMode() != 0)
  {
    previousMillis = currentMillis;

    Serial.println("do something");
  }
}
```

I think it might have to do with the EPROM. Trying to call clearWifiSettings or save did result in the same behaviour.

nrwiersma, 2020-03-14:
Hi,

You are correct. It seems that the EEPROM is not being initialised, even though it should be after `::begin`. It could be hardware that is failing here, have you tired on more than 1 ESP32?

ReneHezser, 2020-03-14:
Hi,
a second one is behaving exactly the same.

nrwiersma, 2020-03-14:
Hi,

I would start debugging be running `EEPROM:begin()` and looking at the bool output, there are logs too. Then run `EEPROM:length()` and check you get the correct size back.

ReneHezser, 2020-03-14:
```c++
  Serial.print("eeprom test");
  bool success = EEPROM.begin(0);
  Serial.print(success);
  success = EEPROM.begin(1);
  Serial.print(success);
  uint16_t l = EEPROM.length();
  Serial.print(l);
```
And the output is:

> eeprom test010

nrwiersma, 2020-03-14:
Ok,

So digging through the ESP32 EEPROM code, i see the issue. ESP32 has gone a a different way when implementing `length` than ESP8266. In ESP8266 `length` tells you the initialised size from `begin`. In ESP32 it has nothing to do with the size in `begin`.

I am not sure how to deal with this.

ReneHezser, 2020-03-14:
When I commented out lines 379-382, it worked for me.
```c++
    // if (EEPROM.length() == 0) {
    //     DebugPrintln(F("WiFi Settings cannot be stored before ConfigManager::begin()"));
    //     return;
    // }
```
How about a variable that keeps track of the initialization through begin()?

nrwiersma, 2020-03-15:
I think i will just disable the check on ESP32. In ESP8266 it is needed to check that EEPROM was actually initialised. Apparently we cannot do that on ESP32.

nrwiersma, 2020-03-18:
Hi,

Can you check if [this branch](https://github.com/nrwiersma/ConfigManager/tree/fix-eeprom-check) works for you. If so I will merge and make a patch release.

ReneHezser, 2020-03-18:
Thank you. Strange the ESP32 does need special treatment. With this change it does work again.

PR #76 (2020-03-18): Dont check EEPROM length on ESP32
Fixes #75
closed 2020-03-18 by ReneHezser