nrwiersma/ConfigManager #81: settings endpoint will only return last saved value

drifterz28, 2020-05-14:
I am not sure if this was a design choice or bug but when saving with a PUT to `/settings` and then calling the GET on `/settings` only the last value is returned. 

I was playing with these 2 functions and I do not know enough about c++ to fix but wanted to bring this up.
https://github.com/nrwiersma/ConfigManager/blob/master/src/ConfigManager.h#L82-L88
and 
https://github.com/nrwiersma/ConfigManager/blob/master/src/ConfigManager.cpp#L177-L194

I confirmed with the save_config_demo example that it does this as it will only show he meta.version (I removed to test the form code)
I also confirmed it will save the values to EPROMM so it's working in that part.

Thanks

nrwiersma, 2020-05-14:
Hi,

Thanks for reporting, that definitely sounds like a bug. Will have a look into it.

nrwiersma, 2020-05-14:
I think i figured out the issue, but am not in a position to test it right now. Would you be able to perhaps if I give you a branch?

nrwiersma, 2020-05-14:
This fix I came up with can be found in #82. I will only be able to test it on a device this weekend. If you can test it before that, that would be awesome.

drifterz28, 2020-05-14:
I can give it a try.

closed 2020-05-15 by nrwiersma

nrwiersma, 2020-05-15:
Release has gone through. Thanks for reporting and your help.

PR #82 (2020-05-15): Fix issue with toJson
This is a potential fix for a toJson issue. 

Fixes #81.