nrwiersma/ConfigManager #84: image file (.jpg/.png) cannot be used in index.html

thanawat14, 2020-06-08:
I try to use an image in html file but the image doesn't show up when in default webpage.

[image] 

PS: I have image.png in data folder.

nrwiersma, 2020-06-09:
ConfigManager does not allow serving arb files from the SPIFFS. This would be a feature in its own right.

cgmckeever, 2020-07-07:
Could you add a [mime/file such as ](https://github.com/nrwiersma/ConfigManager/blob/master/examples/save_config_demo/save_config_demo.ino#L4)
```
const char *imagePNG = (char *)"/image.png";
```

and then 

```
server->on("/image.png", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(imagePNG, mimePNG);
    });
```

you'd hav to define the `mimePNG` I believe at this point as well ...

thanawat14, 2020-07-24:
Sorry for my late reply. I decided to use base64 instead, and the problem was solved. Thank you.

closed 2020-07-24 by nrwiersma