nrwiersma/ConfigManager #97: Problems with with variables in the sketch

Stargazer32584, 2020-11-05:
Hi,

i use example "save_config_demo.ino" from the master branch and have little problems with the variables outside of the struct.

For the float-value in the struct i used

float nameofvariable = config.nameofstructvariable;

So, if i would print it in the serial monitor, He shows only the value 0. I think, the problem in my side.

How can i solve this ?

Another thing is the configuration on the settings site. I write the values in the masks, press "save" and the following screen prints "File not found" message. 

Thank you.

nrwiersma, 2020-11-05:
Hi,

Can you give me a simple code example that demonstrates your issue?

Stargazer32584, 2020-11-06:
Hi,

i only want to serial print the value of "inching delay"

So, i wrote the following additional Code to extract the inching_delay from the struct:

float pause = config.inching_delay;

In the Loop, i use the Serial.print(pause); to print it to the monitor as a test. But, there are only 0. He wrote the value in the memory. But there is no value on the print.

Thank you very much,

André

nrwiersma, 2020-11-06:
Hi,

Can you show me your actual code? Without it will not be able to help you.

Stargazer32584, 2020-11-06:
Alright. I will so it in 4 - 6 hours, because, i'am in work and the code at home :-).

Thanks a lot..

Stargazer32584, 2020-11-06:
Hi,

here is my Code:

#include "ConfigManager.h"

const char *settingsHTML = (char *)"/settings.html";
const char *stylesCSS = (char *)"/styles.css";
const char *mainJS = (char *)"/main.js";

struct Config {
  char device_name[32];
  float inching_delay;
  int8_t led;
  bool enabled;
} config;

struct Metadata {
  int8_t version;
} meta;

ConfigManager configManager;

//Übergabe der Variablen aus dem Config - Manager/////////////////
char* mac = config.device_name;
float delaytime = config.inching_delay;
int led = config.led;

//Millis///
long previousMillis = 0;
long interval = 2000;

void APCallback(WebServer *server) {
    server->on("/styles.css", HTTPMethod::HTTP_GET, [server](){
        configManager.streamFile(stylesCSS, mimeCSS);
    });

    DebugPrintln(F("AP Mode Enabled. You can call other functions that should run after a mode is enabled ... "));
}

void APICallback(WebServer *server) {
  server->on("/disconnect", HTTPMethod::HTTP_GET, [server](){
    configManager.clearWifiSettings(false);
  });

  server->on("/settings.html", HTTPMethod::HTTP_GET, [server](){
    configManager.streamFile(settingsHTML, mimeHTML);
  });

  // NOTE: css/js can be embedded in a single page HTML
  server->on("/styles.css", HTTPMethod::HTTP_GET, [server](){
    configManager.streamFile(stylesCSS, mimeCSS);
  });

  server->on("/main.js", HTTPMethod::HTTP_GET, [server](){
    configManager.streamFile(mainJS, mimeJS);
  });

  server->on("/reboot", HTTPMethod::HTTP_GET, [server](){
    ESP.restart();
  });

}

void setup() {
  Serial.begin(115200);
  DEBUG_MODE = true;
  DebugPrintln(F(""));

  meta.version = 3;

  // Setup config manager
  configManager.setAPName("Demo");
  configManager.setAPFilename("/index.html");

  // Settings variables
  configManager.addParameter("device_name", config.device_name, 32);
  configManager.addParameter("inching_delay", &config.inching_delay);
  configManager.addParameter("led", &config.led);
  configManager.addParameter("enabled", &config.enabled);

  // Meta Settings
  configManager.addParameter("version", &meta.version, get);

  // Init Callbacks
  configManager.setAPCallback(APCallback);
  configManager.setAPICallback(APICallback);

  configManager.begin(config);
}

void loop() {
  
  configManager.loop();   // Um das der Webseitenaufbau der HTML - Datei sauber erfolgen kann, kein delay im Loop einbauen. Besser Millis !!.

  unsigned long currentMillis = millis();

  if(currentMillis - previousMillis > interval) {
        previousMillis = currentMillis;   
  Serial.print(" MAC :");
  Serial.println(mac);
  Serial.println("-------------------------------------");
  Serial.print(" Pause :");
  Serial.println(delaytime);
  Serial.println("-------------------------------------");
  Serial.print("Pausenzeit :");
  Serial.println(led);
  Serial.println("-------------------------------------");
  } 

  // Add your loop code here
}

nrwiersma, 2020-11-07:
So the issue is that `delaytime` is set before the config is read by ConfigManager. Move `float delaytime = config.inching_delay;` to just above the `Serial.println`. To be clear `float delaytime = config.inching_delay;` is taking the current value of `config.inching_delay` and will not be updated as it in not a pointer. This is the source of your issue.

Stargazer32584, 2020-11-07:
Hi,
thank you for your answer !
I put the code directly above the Serial.prints in the loop. Now, the values from mac and led are correct. But the float is 0.00.
Whatever i wrote in the /settings.html in the browser. The value is 0.00 .

That's the only point i had with your tip.

Thanks a lot for spending your time.

Stargazer32584, 2020-11-08:
Hi,

so i tested the code. The problems are homemade and sitting in the front of the monitor #-/.

The code runs perfectly. Thank you very much for your support !

Best regards to South Africa from Löhne, Germany...

André

nrwiersma, 2020-11-08:
Awesome, glad you got it sorted out.

Enjoy,
Nick

closed 2020-11-08 by nrwiersma