nrwiersma/ConfigManager #98: One little problem with one parameter

Stargazer32584, 2020-11-11:
Hi,

my project with an esp32 and a ble - scanner is almost ready to run.

But i have one problem with one parameter. In my case, it is a problem with a char - array from the UI of the ConfigManager.
I convert it to an int with :  int zahl = atoi(rssimac1); .

The BLE - Libary uses a class an in this class, i use an if - request. If the scanner scans a known MAC and the RSSI of it, the esp looks to the rssi value from the ConfigManager and decide " Yes, it is a known adress" or not.

The parameter of the mac are okay. He reads this from the configmanager. But the rssi is again zero.

A serial.print in the loop shows the correct value. That's the problem i have.
Before i used your Manage, i had the INIWifimanager. But it was not mine.
Therefore, i want to use your Libary. The code is experimental and i must clean it up.

Here is my code:

`include "ConfigManager.h"
#include <HTTPClient.h>
#include <WebServer.h>
#include <ArduinoOTA.h>
#include <SPIFFS.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include <BLEAddress.h>
#include <Ticker.h>
#include <NTPClient.h>
#include <WiFiUdp.h>

#define DEBUGGING                     // Einkommentieren für die Serielle Ausgabe

#ifdef DEBUGGING
#define DEBUG_B(...) Serial.begin(__VA_ARGS__)
#define DEBUG_P(...) Serial.println(__VA_ARGS__)
#define DEBUG_F(...) Serial.printf(__VA_ARGS__)
#else
#define DEBUG_B(...)
#define DEBUG_P(...)
#define DEBUG_F(...)
#endif

WebServer server(80);

const char *settingsHTML = (char *)"/settings.html";
const char *stylesCSS = (char *)"/styles.css";
const char *mainJS = (char *)"/main.js";

struct Config {
  char tageins_name[32];
  char tagzwei_name[32];
  char uuideins_name[50];
  char uuidzwei_name[50];
  char rssi_werttageins[4];
  char rssi_werttagzwei[4];
  char rssi_wertibeacon[4];
  char webhook_ein[40];
  char webhook_aus[40];
  int8_t led;
  bool enabled;
} config;

struct Metadata {
  int8_t version;
} meta;

ConfigManager configManager;

///////////////////////////////////////Variablen werden aus dem ConfigManager - struct extrahiert////////////////////////////////
[... 489 lines cut ...]
    delay(100);
    digitalWrite(LEDGruen, HIGH);
    delay(100);
    digitalWrite(LEDGruen, LOW);
    delay(100);
    digitalWrite(LEDRot, HIGH);
    Serial.println("Netzwerk nicht in Reichweite !");
    Serial.println("Verbindung wird aufgebaut...");
    WiFi.reconnect();
    delay(3000);
    digitalWrite(LEDRot, LOW);
    digitalWrite(LEDGruen, HIGH);
  }

  ///////////////////////////////INFO //////////////////////////////////////////

  Serial.print("Aktuelle Minute :");
  Serial.println(timeClient.getMinutes());
  Serial.println("XXXXXXXXXXXXXXXXXXXXX");
  Serial.print("Aktuelle Sekunde :");
  Serial.println(timeClient.getSeconds());

  pBLEScan->clearResults();   // Buffer wird gelöscht, um Speicher frei zu geben

}`

nrwiersma, 2020-11-11:
The problem is the same as last time. When you set a viariable it will not be automatically updated when the config value changes, it takes that value until updated. So when in the start you set `zahl` it is `0` as it should be. It is then never updated. It might be possible that `rssimac1` will be updated as it is a pointer and we use the same memory.

In general it is best to use `config.<something>` directly in your function rather then making global variables.

Stargazer32584, 2020-11-11:
Hi,

thank you for your answer. 
So, you said, it is better to put the `config.<something>` directly in the funktion ?

Is it so right, for example ?

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks
  {
      void onResult(BLEAdvertisedDevice advertisedDevice) // passiert wenn BLE Device ( beacon ) gefunden wurde
      {
        char* rssimac1 = config.rssi_werttageins;
        int zahl = atoi(rssimac1);
        Serial.print("iBeacon gefunden : ");
        Serial.print(advertisedDevice.toString().c_str()); // ibeacon Adresse anzeigen
        Serial.printf("   RSSI: %d ", advertisedDevice.getRSSI());
        Serial.println("---------------------------------");

        if (advertisedDevice.getAddress().equals(*pServerAddress)& advertisedDevice.getRSSI() > zahl)    // ibeacon Adresse Vergleichen
        {
          Serial.println(" Überwachte MAC - Adresse Tag 1 gefunden");         // wenn überwache Adresse gefunden wurde

        }
      }
  };

Thank you...

André

nrwiersma, 2020-11-11:
Yes, that is correct. This will give you the value at the time it is needed.

Stargazer32584, 2020-11-11:
Hi,

you made my day, again :-D !

Thank you very much for your support !!

Best regards and a good evening..

André

nrwiersma, 2020-11-11:
No worries. I will close this issue. Feel free to reopen if needed.

closed 2020-11-11 by nrwiersma