nrwiersma/ConfigManager #99: Is it possible to send a Webpage from ESP32 ?

Stargazer32584, 2020-11-16:
Hi,

the ConfigManager is the perfect libary for me and the projects in the future.

But one question is in my mind. 

From a release before i using the ConfigManager, the ESP32 have send a simple html-site with some values, if i sebd a request to:   http://ESP32IP/site .

But with the ConfigManager, there is no result when he loads the page. 

Is it possible to send this page ?

Best regards

André

nrwiersma, 2020-11-16:
An example for this can be seen here: https://github.com/nrwiersma/ConfigManager/blob/master/examples/custom-http/custom-http.ino#L109

Stargazer32584, 2020-11-17:
Hi,

like this way ? 

`server->on("/site", HTTPMethod::HTTP_GET, [server](){
 server->on("/site", handle_OnConnect);`

But there is no result...

Thanks a lot !

nrwiersma, 2020-11-17:
No, like:

```cpp
const char *siteHTML = (char *)"/site.html";

server->on("/site", HTTPMethod::HTTP_GET, [server](){
    configManager.streamFile(siteHTML, mimeHTML);
});
```

You will need to upload `site.html` into your SPIFFS. You can look to https://github.com/nrwiersma/ConfigManager#additional-files-saved for information.

Stargazer32584, 2020-11-17:
Ah...
But, there is aktually no html-file as site.html. The ESP32 sends it as a String like:

`void handle_OnConnect() {

    server.send(200, "text/html", SendHTML());
  }

  String SendHTML() {
    String ptr = "<!DOCTYPE html> <html>\n";
    ptr += "<head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1.0, user-scalable=no\">\n";
    ptr += "<title>BLEScanner f&uumlr Shelly</title>\n";
    ptr += "<style>html { font-family: Helvetica; display: inline-block; margin: 0px auto; text-align: center;}\n";
    ptr += "body{margin-top: 50px;} h1 {color: #444444;margin: 50px auto 30px;}\n";
    ptr += "p {font-size: 24px;color: #444444;margin-bottom: 10px;}\n";
    ptr += "</style>\n";
    ptr += "</head>\n";
    ptr += "<body>\n";
    ptr += "<div id=\"webpage\">\n";
    ptr += "<h1>BLEScanner f&uumlr Shelly</h1>\n";
    ptr += "<p>Gescanntes Device 1: ";
    ptr += "</div>\n";
    ptr += "</body>\n";
    ptr += "</html>\n";
    return ptr;
  }`

So, he generate the answer from his own code. Is this possible in combination with your libary ?

Thanks..

nrwiersma, 2020-11-17:
```cpp
... config manager setup
configManager.setAPICallback(APICallback);
....

void APCallback(WebServer *server) {
    server->on("/site", HTTPMethod::HTTP_GET, [server](){
        server.send(200, "text/html", SendHTML());
    });
}
```

Stargazer32584, 2020-11-17:
Tried it.
The following compilation error appears:

"request for member 'send' in 'server', which is of pointer type 'WebServer* const' (maybe you meant to use '->' ?)

Then, i switched to:

`server->on("/site", HTTPMethod::HTTP_GET, [server](){
  server->send(200, "text/html", SendHTML());
  });`

..and what should i say....?

Thanks again, man. It works...!!

Thanks a lot and best regards...

André

nrwiersma, 2020-11-17:
No problem

closed 2020-11-17 by nrwiersma