nrwiersma/ConfigManager #100: Unable to save WiFi settings to EEPROM [esp32]

9ae8sdf76, 2021-02-12:
Hi, I came across your library on platformio, and I'm excited at what it has to offer. After following the example in the README, I get the following output and my MCU stays in AP mode waiting to be set up.

```
Reading saved configuration
MAC: 7C:9E:BD:ED:28:5C
SSID: "��������������������������������"
MagicBytes mismatch - SSID: ������������������������������������������������������������������������������������������������Xvǒ
Starting Access Point
E (2364) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

AP Name: Desk-Buddy
AP IP address: 192.168.1.1
Webserver enabled on port: 80
[E][WebServer.cpp:617] _handleRequest(): request handler not found
Unknown URL: captive.apple.com
[E][WebServer.cpp:617] _handleRequest(): request handler not found
Unknown URL: captive.apple.com
[E][WebServer.cpp:617] _handleRequest(): request handler not found
Unknown URL: captive.apple.com
[E][WebServer.cpp:617] _handleRequest(): request handler not found
Unknown URL: captive.apple.com
Storing WiFi Settings for SSID: "---------------"
[E][EEPROM.cpp:199] commit(): error in write
EEPROM committed: false
```

For good measure, I did run `~/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool --chip esp32 erase_flash` to make sure that my flash was clean, and I still get the above messages. I am not sure how to proceed, and could use your advice. 

Thank you.

nrwiersma, 2021-02-12:
Hi,

It seems there is an issue with your version of Arduino for ESP32. It is not handling requests or able to commit data to the EEPROM, so this lib will never work. I suggest you play with different versions of Arduino for ESP32 and figure out at which version it starts/stops working, as well as produce some sample code.

9ae8sdf76, 2021-02-12:
That's a shame. I don't use Arduino directly; just the framework through PlatformIO. I can't help but suspect something else given the garbage in the SSID text and the "MagicBytes mismatch", but I do not have the free cycles or knowledge on how to troubleshoot it either way. Thank you for your time. It's back to the drawing board for me.

closed 2021-02-12 by 9ae8sdf76

9ae8sdf76, 2021-02-13:
Ok, so I couldn't help but do some more reading about EEPROM, and what wasn't obvious to me was that there is a 4k limit (not sure if it's board-specific or not) on the amount of data that can be stored. Once I brought my storage needs down to below 3k everything worked as expected. It might be worth putting some sort of warning or debug statement about the size of the config in case someone else makes the same mistake. :)

/cc @nrwiersma