nrwiersma/ConfigManager #101: Saving char arrays in EEPROM

dezur, 2021-02-12:
Hi! First of all thanks for this great library! Yesterday I have faced an issue when I was trying to store char arrays. 

So I made a struct like this:
`struct Config {
  float nilBitcoin;
  float nilEthereum;
  float zolBitcoin;
  float zolEthereum;
  char linehe1[10];
  char linehe2[10];
  char linehe3[10];
  char linehe4[10];
} config;`

Floats are stored correctly, but the char arrays not. I am getting garbage when I try to load.

Solution: I have to move char arrays to the top of the struct:
`struct Config {
  char linehe1[10];
  char linehe2[10];
  char linehe3[10];
  char linehe4[10];
  float nilBitcoin;
  float nilEthereum;
  float zolBitcoin;
  float zolEthereum;
} config;`

Is it a bug in your library or in the ESP32 framework or my knowledge about structs and EEPROM store is need to be updated? :)

nrwiersma, 2021-02-13:
I would need to see how ConfigManager is setup with the config.

nrwiersma, 2021-08-29:
Considering stale and closing. Feel free to reopen.

closed 2021-08-29 by nrwiersma