nrwiersma/ConfigManager #109: Metadata adding config values

andyb2000, 2021-06-25:
Hi,
Perhaps I'm a little confused to the point of Metadata as opposed to Config?
I'm using Config to retain the wifi settings, then was going to use the Metadata struct to contain my user parameters, is this right or not? Does it get saved, and if so do I just call:
  configManager.save();
and that will store the meta and config settings?
The settings don't appear to save for the meta part of the code, is this expected behaviour and if so what is the key point of the meta part of this and examples?

My code:

`include <ConfigManager.h>
ConfigManager configManager;
struct Config {
    char name[20];
    bool enabled;
    int8 hour;
    char password[20];
} config;

struct Metadata {
    int8_t version;
    String setting_autoupdate;
} meta;

setup() {
  configManager.setAPFilename("/ap-index.html");
  configManager.addParameter("name", config.name, 20);
  configManager.addParameter("enabled", &config.enabled);
  configManager.addParameter("hour", &config.hour);
  configManager.addParameter("password", config.password, 20, set);
  configManager.addParameter("version", &meta.version, get);
  configManager.addParameter("setting_autoupdate", &meta.setting_autoupdate, set);
  configManager.begin(config);
};

loop() {
 meta.setting_autoupdate="yes";
 configManager.save();
`

nrwiersma, 2021-06-25:
Hi,

You are correct that metadata does not get saved. The point behind metadata is to expose static information via the API, for example API version.

andyb2000, 2021-06-25:
ah understood, so moving my parameters into the struct Config is the correct implementation.
Thank you.

closed 2021-06-25 by andyb2000