nrwiersma/ConfigManager #110: reading and writing from struct config data

andyb2000, 2021-06-25:
Hi,
I'm trying to read and write from the config structured data. Am I missing something in the implementation here.
Please see example code below:

`
#include <ConfigManager.h>
ConfigManager configManager;
struct Config {
    char name[20];
    char setting_autoupdate[1];
    char setting_mqttdest[15];
    bool enabled;
    int8 hour;
    char password[20];
} config;

struct Metadata {
    int8_t version;
} meta;

setup() {
configManager.setAPFilename("/ap-index.html");
  configManager.addParameter("name", config.name, 20);
  configManager.addParameter("enabled", &config.enabled);
  configManager.addParameter("hour", &config.hour);
  configManager.addParameter("password", config.password, 20, set);
  configManager.addParameter("version", &meta.version, get);
  configManager.addParameter("setting_autoupdate", config.setting_autoupdate, 1);
  configManager.addParameter("setting_mqttdest", config.setting_mqttdest, 15);
  configManager.begin(config);

  if (strlen(config.setting_autoupdate) == 0) {strncpy(config.setting_autoupdate, "0",1);};
  if (strlen(config.setting_mqttdest) == 0) {strncpy(config.setting_mqttdest, "0.0.0.0",7);
     Serial.println("Settings defaults as meta data empty");
  } else {
     Serial.println("Reading settings from meta data values");
  };
configManager.save();
}

loop() {
DEBUG_PRINT("configmanager mqttdest test: " + String(config.setting_mqttdest));
}
`

The results are, on running the script it shows "Settings defaults as meta data empty", so it's set the defaults as the config.setting_mqttdest length is 0.
It should then save this value.
The loop runs and correctly then shows the value contains 0.0.0.0 as expected.

I then reload the esp8266 so the code restarts from blank.
It does the same, no values stored so it says "Settings defaults as meta data empty" and sets the default value of 0.0.0.0 again.
It never seems to write the values and is blank at each reboot.
What am I doing wrong?

So to write I am doing:
`
strncpy(config.setting_mqttdest, "0.0.0.0",7);
configManager.save();
`
That is correct?
Thank you!

nrwiersma, 2021-06-28:
Hi,

The code looks correct from what I can see. ConfigManager is using the EEPROM to store between reboots, so I would suggest you run some simple EEPROM tests to check if it is the board.

andyb2000, 2021-06-28:
Thank you, I've found the issue, corrupted eeprom it seems so clearing has solved it.

closed 2021-06-28 by andyb2000