nrwiersma/ConfigManager #115: CORS errors

Frtrillo, 2021-11-02:
[image]

How can we set the webserver to ignore cors policy?

nrwiersma, 2021-11-03:
You would need to add `server.sendHeader("Access-Control-Allow-Origin", "*");` to all `handle*` functions in the lib. A pull request to this effect would be welcome.

Frtrillo, 2021-11-03:
> You would need to add `server.sendHeader("Access-Control-Allow-Origin", "*");` to all `handle*` functions in the lib. A pull request to this effect would be welcome.

Im willing to do it but it still isn't working. I've added it like this:
[image]
And now im getting a different error:

[image]

nrwiersma, 2021-11-03:
Chrome does not support CORS on localhost. That is the reason for the other issue.

Frtrillo, 2021-11-03:
> Chrome does not support CORS on localhost. That is the reason for the other issue.
Angular app will do the request from localhost though, which is one of the mainly (and good) uses for the library I think. Somehow throught browser is working /scan:
[image]
And im getting the networks json. But it isn't working on the phone indeed.

What I still can't make it to work is to POST:
[image]

Maybe adding something like send status 200 would work?

nrwiersma, 2021-11-03:
The issue is that your server is on localhost, and chrome has blocked CORS on localhost: https://stackoverflow.com/questions/10883211/why-does-my-http-localhost-cors-origin-not-work

Frtrillo, 2021-11-04:
> The issue is that your server is on localhost, and chrome has blocked CORS on localhost: https://stackoverflow.com/questions/10883211/why-does-my-http-localhost-cors-origin-not-work

I've fixed it to work on the localhost browser too, Ill do a pull request later

PR #116 (2021-11-04): Fixed CORs and POST issues on localhost and browser
Added CORS request to work on localhost chrome and added http options answer to POST requests 

Fixes #115
closed 2021-11-04 by nrwiersma