nrwiersma/ConfigManager #117: How can I make it to auto reconnect?

Frtrillo, 2021-11-05:
Im trying to make the library (or sketch) to auto reconnect wifi after it connected successfully. Because right now once its connected and if wifi stops working and works again, it won't reconnect and connexions wont work.

I've tried something like this modifiying the library but its not working:

nrwiersma, 2021-11-06:
You should try calling `WiFi.reconnect()`.

Frtrillo, 2021-11-06:
> You should try calling `WiFi.reconnect()`.

Tried that first on the ::loop method and didn't worked

cgmckeever, 2021-11-06:
you might need to call `ConfigManager::wifiConnected()` (or similiar) to first check status and the reconnect .. or it may perpetually do some reconnect on/off thing

Frtrillo, 2021-11-10:
> > You should try calling `WiFi.reconnect()`.
> 
> Tried that first on the ::loop method and didn't worked

Using it inside the sketch instead have worked

closed 2021-11-10 by Frtrillo