nrwiersma/ConfigManager #122: Compiling / version issue with ArduinoJson

bmw-biker, 2023-03-30:
Hi, I tried to get ConfigManager running. But I get lots of errors regarding ArduinoJson getMember(), getOrAddMember(), .is<T> and .as<T>.
Obviously Platformio takes the latest ArduinoJson 6.21.1.
Seems that between 6.0.0 and the current there have been some breaking changes.

Any idea how to adapt ConfigManager to current ArduinoJson ?
Or, which ArduinoJson version is the correct for latest ConfigManager, and how can I force Platformio to use this version ?

Example log for getMember and .is<T>:
```
.pio\libdeps\esp_wroom_02\ConfigManager\src/ConfigManager.h: In member function 'void ConfigParameter<T>::fromJson(ArduinoJson::JsonObject*)':
.pio\libdeps\esp_wroom_02\ConfigManager\src/ConfigManager.h:78:42: error: 'class ArduinoJson::JsonObject' has no member named 'getMember'
   78 |     if (json->containsKey(name) && json->getMember(name).is<T>()) {
      |                                          ^~~~~~~~~
.pio\libdeps\esp_wroom_02\ConfigManager\src/ConfigManager.h:78:62: error: expected primary-expression before '>' token
   78 |     if (json->containsKey(name) && json->getMember(name).is<T>()) {
      |                                                              ^
```
Best regards, Torsten

nrwiersma, 2023-03-30:
Sorry, have no idea. If you figure it out, feel free to open a PR.

bmw-biker, 2023-03-30:
Many thanks for the fast response.

Meanwhile I figured some out...
Latest version of ConfigManager is from 12.2021.
So I tried to find the latest ArduinoJson version before that date which is 6.18.5.
So, if I then include the following exact version 6.18.5 dependency in platformio.ini file it compiles without error:
```
lib_deps = 
  bblanchon/ArduinoJson@6.18.5  ; without ^ before version !
```

Although I didn't manage to get it to connect with my Smartphone yet. Smartphone tries to connect but aborts after a while. I'll have to check tomorrow.

nrwiersma, 2024-05-08:
Fixed by #123. If there are still issues, please reopen.

closed 2024-05-08 by nrwiersma